heszegi/com.heszi.deye #1: Device not found by the app

MarceloSilvaLeite, 2024-08-27:
I can't add my system to Homey via the Deye app. I get the following message:
No devices have been found.

Diagnostics Report:
6e878e44-6ab7-4176-a1ea-963a0fb68277

heszegi, 2024-08-27:
Hi Marcelo!

It looks like there is no station linked to your account. Do you use the same login/pass that you use in DeyeCloud application or web app? Do you see your station there?

And could you please tell where your station registered? Currently the app uses the EU server hardcoded, maybe this may cause some problems.

If your devices are in Europe, Africa, or the Asia-Pacific region: Europe data center.
If your devices are in North or South America: Americas data center.

Selection for these not implemented yet.

MarceloSilvaLeite, 2024-08-28:
Hello!

Yes, I use the same login/password that I use in the DeyeCloud app.

Yes, I see my station there.

My station is registered in the Americas data center.

Do you have an estimate of when the selection for the Americas data center will be implemented?

heszegi, 2024-08-28:
I have no strict timeline for this project, but I will prioritise this functionality, and let you know if done.

heszegi, 2024-08-28:
Hey Marcelo, please try the latest test version, it has the data center selection functionality: https://homey.app/a/com.heszi.deye/test/

MarceloSilvaLeite, 2024-08-28:
Yes, I have now managed to add my station. Great job.

closed 2024-08-29 by heszegi