heszegi/com.heszi.deye #2: App does not show serial number. Request for daily production information

MarceloSilvaLeite, 2024-09-01:
Hello, my Deye system does not display the serial number as it should.
Is it possible to implement daily production in the app? I have a Fronius system and it reports daily production, this would be very useful.

Diagnostics Report:
a2eee962-39c0-4a44-864b-153a75ddfb20

heszegi, 2024-09-01:
This looks an interesting issue with the DeyeCloud API.

According to the API, it appears that no inverters are added to your station, but I believe this is not the case. I suspect that the application I generated in the DeyeCloud system may not have been granted the necessary permissions. If you look at the two attached photos, you’ll notice that the 'AMEA' application doesn’t list any permissions under 'Access Control', while the 'EMEA, APAC' application has all the required permissions. However, I created both applications in the same way, requesting all permissions.

[image]
[image]

My theory is that since my account is registered in the EU region, I may be unable to generate an app with the necessary permissions for the 'AMEA' region.

Could I ask you to try creating an app at you own region? If you could send me the required information, I’ll build the Homey app for you to test.

Here’s what you need to do to help test this:

- Visit DeyeCloud Developer Portal: [https://developer.deyecloud.com/](https://developer.deyecloud.com/)
- Click "Sign In" at the top right corner.
- Choose the 'AMEA' region.
- Log in with your DeyeCloud credentials.
- Allow access to "use your account and avatar."
- Click on "Explore API."
- In the left menu, select "Application."
- Generate an application and make sure to select all three 'Access Control' checkboxes.
- Email me the AppID and AppSecret at heszegi@gmail.com.

I hope this process is straightforward, but please let me know if you need any assistance.

Thank you for your help!

heszegi, 2024-09-03:
It seems that DeyeCloud is having trouble creating an app with the necessary access rights in the AMEA region. We've reached out to Deye support to learn more about the issue.

closed 2024-09-03 by heszegi