heszegi/com.heszi.deye #3: Change settings in inverter

Tooltim73, 2024-09-05:
Hello, Thanks for your great app. I was wondering if there is a way to control  the inverter, as in change any setting so i can choose in homey to switch on/off the Grid Charge in a specific time period (time of use) because so i can charge the battery when there is no sun predicted for the next day.

thanks

heszegi, 2024-09-05:
Hello Tooltip73, thank you! I'm glad to hear the app is helpful to others besides me. Yes, updating the app to include editing settings available through the DeyeCloud API is on the roadmap. However, I can't provide an estimate for when that update will be released just yet. But I'm working on it.

closed 2024-09-05 by heszegi

Tooltim73, 2024-09-05:
Thanks that would be perfect