heszegi/com.heszi.deye #4: Missing feature

Botje1987, 2024-10-09:
Nice App Lots of information
Only missing few things.

Is not ot showend in the energy app as consumption and usage energy on the energy platform . 
Missing the messing option when grid is missing 

hoop if it is possible to make these things possible 

yours sincerely 

Leon Bot

heszegi, 2024-10-09:
Hi Leon,

Thank you for your ideas, these are great features.

Could you explain a bit more about how you want this device to be integrated into the energy platform? I'm not very familiar with that part of Homey, so I'm having a hard time imagining how to implement it.

As for the grid status, it's currently not supported by the Deye API. However, they told to plan to add new features in the future, so we're hopeful it will be included.

bbrodnik, 2024-10-29:
I'm very gratefull for this application.

If possible, I woud welcome two features I miss very muuch:

1. Reading and enabling/disabling Peak Shaving option. It significantly changes selling first mode functionality of inverter, so possibibility of changing this option would help very much in optimizing battery charge/discharge functionality in selling first mode.

2. Grid missing information would enable to react in glid blackout events to notify users and disable unnessesary loads if battery goes too low. And also to manage thing when grid comes back.

Regards,
Brane

heszegi, 2024-11-02:
Hi @bbrodnik, thank you for your message! Your ideas are great. Unfortunately, the 'Peak Shaving' option isn’t supported by the Deye API, so we can’t implement it until they add that functionality—hopefully, they will. As for the Grid Availability feature, it’s also not currently supported by the API, but I’m testing a solution that might enable it. Hopefully, it’ll be available soon in the Homey app.

heszegi, 2024-11-06:
App v1.2.0 available with Grid availability function for testing https://homey.app/a/com.heszi.deye/test/

closed 2024-11-06 by heszegi

bbrodnik, 2024-11-07:
Great, I tested and it works well, both #4 and #6.

 

Thank, you,

Brane

 

Closed #4 <https://github.com/heszegi/com.heszi.deye/issues/4>  as completed.