heszegi/com.heszi.deye #5: Homey Pro 2023 and Deye Station

Jcbnc, 2024-10-25:
Hello, Deye station was showing fine in my devices for the last two weeks and this morning it is showing as unreachable. I tried to remove the device and the Homey app then reinstall but the device is still unavailable. Please advise. With best regards.

heszegi, 2024-10-25:
Hi @Jcbnc, I see what you mean. It’s a global issue, DeyeCloud appears to have stopped responding. I’ll look into what might have happened.

heszegi, 2024-10-25:
@Jcbnc, Please try using the Repair option and log in again with your credentials. It should work now. Let me know the result.

Jcbnc, 2024-10-25:
Thank you for your prompt reply, it is working again now.Many thanks :-)Jean-Claude ***@***.***: +687 542239Pensez à l'environnement - est-il indispensable d'imprimer ce courriel ?Please consider the environment - do you really need to print this email ?

closed 2024-10-25 by heszegi