heszegi/com.heszi.deye #6: Set Battery Max Charge Current can't be set to 0 anymore

bbrodnik, 2024-11-05:
A have Set Battery Max Charge Current card in flow for a while, but after some other changes in flow I can't save it any more due to Invalid arguments error in mentioned card. I don't understand why since it worked, until last update, I think.
[image]

heszegi, 2024-11-05:
I'm sorry to hear about this issue. Unfortunately, I'm not sure why it happened, but I can confirm that the minimum value for this setting is 1 and hasn’t changed since this card added in the previous version of the app. It’s likely that an error occurred in Homey base system during the app update, causing the value to reset to 0. Please try deleting the card and adding it again with a valid value. 

Please let me know the result

bbrodnik, 2024-11-05:
The minimum value for this setting should be 0 to disable charging, not 1.  It was intentionally set to zero and it worked.
It doesn't matter if a delete and add the card.

heszegi, 2024-11-05:
Hmm... that's really interesting! The minimum value has always been set to 1 in the configuration files for this card, so it must have been a bug that allowed you to set it to 0.

Your solution to disable charging/discharging sounds intriguing, though. I’ll look into it, and if it doesn’t cause any issues, I'll consider allowing a 0 setting in a future minor release.

bbrodnik, 2024-11-06:
Manual input on inverter allows 0, so why API shouldn't?

closed 2024-11-06 by heszegi