heszegi/com.heszi.deye #7: Deye App gives erroneous information after the upgrade to version 1.1.0

damianarielmauro, 2024-11-05:
After upgrading the Deye App to version 1.0.0, the application gives erroneous information. 
In my case, my plant in Deye Cloud has 2 inverters and before the last upgrade of the App the Deye device in Homey showed the sum of both inverters for each parameter (Power from Grid, Consumption, Power from Cells, Power from Battery, etc.), but after yesterday upgrade now it shows the values ​​of the parameters of only the Master inverter.
This is affecting all the automation that takes place with the Flows already configured !!!

Please fix it soon!

Thanks

damianarielmauro, 2024-11-05:
Correction:
I meant to say: "After upgrading the Deye App to version 1.1.0..."
It was after the upgrade of Nov, 3rd.

Thanks a lot for a prompt fix!

Thanks
Regards

heszegi, 2024-11-05:
Hi @damianarielmauro,

The issue you're experiencing is a known one. The new data source I’m using only provides device-specific data rather than station data, but it offers more detailed information directly from the device. This allows for more helpful functionality for most users, especially those with a single main inverter in a station.

I’m working on a solution for your setup but need a bit more information about your station (multi-inverter setup). I've created a test version (v1.1.1: https://homey.app/a/com.heszi.deye/test/)  of the app—please update it and send me a diagnostics report to help me find a suitable solution.

Thank you,
Heszi

damianarielmauro, 2024-11-06:
Hi Heszi,

First of all, your app is great!

It was working perfectly for me until yesterday's upgrade.
Deye Cloud allows you to view the parameters for each inverter individually but can also show the parameters for the plant as a whole by adding up the individual parameters for each inverter.

My installation currently has 2 inverters, but next week I'm actually going to add a third one.
Please take a look to the pictures attached with the information showe in Deye Cloud.
The app is now showing the values for inverter 2305196459 only.

I'm going to try out the new test version v1.1.1 and let you know how it works.

Thanks a lot!
Regards

[image]
[image]
[image]

damianarielmauro, 2024-11-06:
I've updated and tested v1.1.1 and it is still showing info of one inverter only.
Diagnostic Report:
44eccd86-77b5-4999-9812-c4af1abd07d5

Thanks

damianarielmauro, 2024-11-06:
Hi @heszegi 

I've updated and tested v1.1.1 and it is still showing info of one inverter only.
Diagnostic Report:
44eccd86-77b5-4999-9812-c4af1abd07d5

Thanks

heszegi, 2024-11-06:
Thank @damianarielmauro for the detailed information and the diagnostics report. I can now see the structure of your multi-inverter system.

This latest patch wasn't intended to solve your issue directly—it was primarily to log the information needed to better understand your setup.

Unfortunately, handling multiple devices together in Homey works a bit differently than in the Deye app, so managing them isn’t straightforward. I have some ideas on how to address this, but it will require a more extensive refactor of the application.

Resolving this issue is now a high priority on our roadmap. Please bear with us, and I'll keep you updated once the refactor is complete and the app is ready to support this type of system.

damianarielmauro, 2024-11-06:
Hi @heszegi 

If you think this might take some time, would it be possible for you to publish the previous v1.0.0 version that worked fine with my system, just like you released v1.1.1, just temporarily? 

I have flows that perform daily power management automation that I could continue to use in the meantime until I get the new version.

Thanks in advance for the support.

heszegi, 2024-11-06:
You can download version 1.1.2 of the app here: https://homey.app/en-hu/app/com.heszi.deye/Deye-Solar-Systems/test/

This version is the same as 1.0.0 was.

After installing it, please let me know and disabel auto-updates for Deye app in your Homey settings, as I'll be introducing version 1.2.0 with a new feature soon, but it will also disrupt your setup. 

Please only enabled auto-update only when you see "Multi-Inverter Station Support" mentioned in the changelog in the future.

damianarielmauro, 2024-11-06:
Hi @heszegi
I installed version 1.1.2 and it is working like before showing total values of parameters adding from two inverters.
Thanks!

Anyway I will be monitoring when the new versions mention "Multi-Inverter Station Support" because I'd like to have the new features too (in special for "Grid Availability" about Grid missing information that would enable to react in grid blackout events, and I was thinking to ask you to add also the option to enable/disable the "Time of Use" under System Work Mode-2). 

Thanks for the app, it is great!
Regards

closed 2024-11-06 by heszegi

damianarielmauro, 2024-11-27:
Hi @heszegi,

I just wanted to ask you if the new version already has "Multi-Inverter Station Support"?
I would like to be able to use some features of the new version like the "Grid Available" card and others, but without disrupting my setup that has Multi-Inverter.

Thanks for your development of this app!

heszegi, 2024-11-27:
Hi @damianarielmauro,

The answer is both yes and no. In the new version, you can change the data source, so updating is an option. You can select your preferred data source in the Settings menu, and in your case, "Station" would be the best choice. However, switching to "Station" will disable the "Grid available" trigger.

Essentially, this doesn’t improve your current situation. The issue lies with the Deye API, it only provides the needed grid data if you use the device as the data source. But in your case, with two inverters, it will only display data for the first device for now.

So yes, you can update and maintain the same functionality as now.

I hope to find some time over the Christmas vacation to work on a better solution for multi-inverter support.

damianarielmauro, 2024-11-27:
Hi @heszegi

So for now I will stay on the previous version without enabling automatic upgrade.
If you don't mind, I'll check back with you in a few weeks... next year :)

Merry Christmas! 
Thanks