heszegi/com.heszi.deye #8: Unable to change settings on Inverter

HelgeJr, 2024-12-08:
Hi, I have been able to connect the App with my DeyeCloud Account, and also to read the data from the station when selecting Station as the Data Source. But when selectiong Station as Data Source it claims in the Homey display that "No Device Found"

What I have not been able to work though is:

- I am unable to get any data from the app, when trying to connect to Device as Data Source, it states that the unit is not available 
- I have not been able to change settings on the inverter (Changing value on Grid Charge) when connected with Station as Data Source

[image]
[image]
[image]

Diagnostic Report : 358ab6c0-84a0-4bad-ab8d-e5d6d3f0ba65

Please advise. With best regards.

heszegi, 2024-12-09:
Hi @HelgeJr ,

Please try removing and re-adding the device, making sure to select the correct Data Center during the setup. Let me know the result. Initially, test both data sources, if the device isn’t working, there’s no need to try the device setting as it won’t work too.

HelgeJr, 2024-12-10:
Thanks @heszegi ,

I tried to remove and re-adding the device, and as far as I can remember I did exactly the same as the first time, but luckily this time it actually came up with the Inverter Serial Id immediately, and I am also able to make changes to the setting.

So no idea what I did differently the first time, but next time I have problems I will for sure try to reinastall the App before bothering you :)

So thank you for the help, and I have been waiting for the possibility to control the Inverter with more logic then on the basic installation , so really happy for the effort you have put into this App.

Best regards
Helge

closed 2024-12-10 by HelgeJr