heszegi/com.heszi.deye #10: Setting Grid Peak Shaving parameters from app

bbrodnik, 2024-12-30:
Thank you for aplication you provided to us.

We have tariff system where peak power is charged differently during the day.  As Time of use table in inverter doesn't support grid peak shaving settings, I'm seeking for this possibility through your Deye Homey app.

My question: would it be possible to enable/disable grid peak shaving and to set up grid peak shaving power from app?

heszegi, 2024-12-31:
Hi @bbrodnik the Grid Peak Shaving settings are not available over the Deye Cloud API (https://developer.deyecloud.com/api), so currently this app not able to set it.
Sorry Heszi

bbrodnik, 2024-12-31:
But is there any chance for Grid Peak Shaving settings to be added to over the Deye Cloud API in near future?

What is the protocol to do so?

 

Regards,

Brane

heszegi, 2025-01-05:
There’s no protocol for this yet. I’m completely independent from Deye (not sponsored or supported by them) and have no visibility into their roadmap, timeline, or plans for API development. I originally designed this app to monitor solar energy generation and promote smart energy consumption. Managing or deep handling the inverter directly through the app could be risky, so it wasn’t part of my initial plans.

bbrodnik, 2025-02-21:
> Hello!
>
>
>
> Any positive news on the matter? Sun is getting stronger and switching peak shaving option through cloud API is becoming more and more beneficial to overcome inverter issues.
>
>
>
> Please let me know if any information available.
>
>
>
> Best Regards,
>
> Brane Brodnik
>
>
>
>
>
>
>
>
>
>
> [... 82 lines cut ...]
>
>
> Regards,
>
> Brane Brodnik

Hi, I understand you and appreciate your work very much. And together we can achieve more.

 

I kindly asked Deye Cloud API Support to include grid shaving parameter settings into their API and was successful.

I kindly ask you now, and it would be great, if possible, to include peak shaving control into your Homey app.

 

Please, let mi know if it would be possible.

Best regards,

Brane Brodnik

 

Here is Deye info:

 

Hello 

 

GridPeakShaving control is ready, please refer to endpoint doc on Developer Portal.

 

 

 

Yunqing Zhang

bbrodnik, 2025-02-21:
[image]

heszegi, 2025-02-23:
You are welcome :)

closed 2025-02-23 by heszegi