heszegi/com.heszi.deye #11: time of use settings

munksbo, 2025-01-29:
will it be possible to include the option of being able to set and update time of use settings. in this way, you will be able to smartly choose when the inverter should charge the batteries, in relation to when the electricity is cheap

[image]

closed 2025-01-29 by munksbo

bbrodnik, 2025-03-10:
I would also be very happy if Time Of Use table settings would be implemented: [/v1.0/order/sys/tou/update].

It would allow changing TOU table on daily basis. For example to discharge batteries deeper before sunrise on clear sky weather report. Or to retain more energy in batteries for peek shaving on cloudy or rainy weather report.

I understand that filling full array of paramters is challenging, but it could be simplified. 
Each request could only change one set of parameters, addressed by index in range between 0 and 5.

Missing sets could be acquired from API [/v1.0/config/tou].

Would it be possible somehow to implement it?

Thank you.

bbrodnik, 2025-03-12:
> Hi @bbrodnik <https://github.com/bbrodnik> ,
>
> I'm not planning to add the 'Time Of Use' functionality in the near future, as I want to keep the app lightweight and as simple as possible. Additionally, this feature is not essential for most users.
>
> However, you might consider switching to the BlauHoff app (https://homey.app/en-hu/app/com.sortedbits.blauhoff/BlauHoff/), which also supports Deye inverters via the Solarman protocol with local data updates (which is also the next step in my app development roadmap) and offers more robust capabilities as well.
>
> I hope this help,
> Best
> Heszi
>
>
>
> Hi @bbrodnik <https://github.com/bbrodnik> ,
>
> I'm not planning to add the 'Time Of Use' functionality in the near future, as I want to keep the app lightweight and as simple as possible. Additionally, this feature is not essential for most users.
>
> However, you might consider switching to the BlauHoff app (https://homey.app/en-hu/app/com.sortedbits.blauhoff/BlauHoff/), which also supports Deye inverters via the Solarman protocol with local data updates (which is also the next step in my app development roadmap) and offers more robust capabilities as well.
>
> I hope this help,
> Best
> Heszi

Thank you for info. I tried BlauHoff app and it’s versatile, but I probably faced compatibility issues with my inverter (Deye SUN-30K-SG01HP3-EU-BM3) which is not supported, although most of values seems to be shown correctly with SUN-*K-SG01HP3-EU-AM2 configuration. Refresh rates are much better, but setting Time of Use slots unfortunately doesn’t work for me. 

 

Best regards,

Brane