heszegi/com.heszi.deye #12: Possible updates

AndreasEklind, 2025-02-19:
Hi,

First of all, THank you for this app! Finally I can start using my Deye setup in my smart home :)

I would like to be able to setup something like this:

WHEN
(Homewizzard)
"current usage is over 500w"
AND
(Deye Solar Systems)
"Battery Max Discharge Current is 5A"
THEN
(Deye Solar Systems)
"Set Battery Max Discharge Current to 10A"

I hope you understand what I´m after. I´m pretty new to this.

I would like to be able to set the discharge to follow my usage

Thank you for your great work! :)

heszegi, 2025-02-19:
Hi @AndreasEklind , you're welcome! Currently, the app cannot read settings data from the inverter or station because the open Deye Cloud API does not support it. This means you can't create the exact flow you described. However, if you set up this flow without the "And" card, the final outcome will essentially be the same. Alternatively, you can store the set Current value in a variable in Homey, allowing you to check the actual value there while still setting it exclusively through Homey.

Meanwhile, keep an eye on app updates, as I’m working on a solution that will enable direct communication with the inverter without relying on the Deye Cloud. This will open up new possibilities, but at the moment, I don’t have a timeline for when the new version will be available.

closed 2025-02-19 by heszegi

AndreasEklind, 2025-03-10:
Thank you for your reply!
That update sounds amazing! Can't wait til it's done!
Your work is highly appreciated!

AndreasEklind, 2025-04-19:
Hello again!

Is it possible to add the info from my inverter to the new energy tab? I
can see it but it don´t show any numbers. See attached.
Have I missed a setting?

heszegi, 2025-04-22:
Hi Andreas,

Yes, I do plan to add that feature, but I don’t have a timeline for it at the moment. I’m currently focused on some other projects.

Best
Heszi