heszegi/com.heszi.deye #14: No new devices have been found

horatiuanghel, 2025-03-26:
Hello!

I have a Homey Pro and one Deye SUN-8K-SG04LP3-EU. 
When I try to add the device into Homey app, after login I am receiving the message "No new devices have been found" and a Close button.
Do you have any ideea what could be the problem ?

Thank you!

[image]

heszegi, 2025-03-27:
Hi @horatiuanghel,

These are the possible cases:

- The device is already added.
- The wrong data center location was selected.
- There is an error in Deye Cloud device registration (check the official Deye Cloud).
- The device is not supported (although your device should theoretically be supported).

Best regards 
Heszi

horatiuanghel, 2025-03-27:
Hello,

Problem solved. 
The account should be for end user. 

Thank you!

closed 2025-03-27 by heszegi