heszegi/com.heszi.deye #16: Solar generation not displayed despite proper export data

Iupiter1979, 2025-05-29:
Hello,

Thanks a lot for the app, we Deye users really needed something like this. If there’s a way to support the project financially via PayPal, I’d be happy to contribute.

I’ve installed a 10 kW single-phase hybrid Deye inverter to replace my 6 kW Solis after expanding my system. The app connects to DeyeCloud just fine, but the solar generation values don’t seem to show up correctly. It always displays zero production, although Deyecloud shows the values just right.

**What’s interesting is that the grid and battery export data is shown properly** (using Deye’s convention of a negative sign to indicate power being sent to the grid). So it’s clear that solar is being generated, and if you add up the values, they match exactly what DeyeCloud reports as total production.

The main issue is that from Homey’s Energy Manager point of view, there’s no solar generation shown at all,  even when a large amount is clearly being injected into the grid. That also makes it harder to build flows based on solar production.

Have you seen this happen before, or could it be something specific to my setup?

[image]<br>
[image]<br>

[image]

heszegi, 2025-05-29:
Hi @Iupiter1979,

Thanks again for reaching out, and for your kind words and offer to donate! That really means a lot.

The good news is that your system is working just fine. The issue you're seeing comes from the Homey Deye app itself, it was originally created before Homey's Energy tab feature existed. This Energy tab is a relatively new feature (only available for a few months), which is why the app doesn't currently support it.

Adding full support for the Energy tab is definitely something I plan to do, but at the moment I just don’t have the free time to work on it. Hopefully, I’ll be able to tackle it later this year.

You're right that not having direct solar production in the Energy tab can make it harder to build flows. However, with Advanced Flows and variables, it’s still possible to create logic based on solar production, basically everything can still be solved that way, even if it requires a bit more setup.

As for your generous offer to contribute financially: I really appreciate it, but for me, this is just a hobby project — it’s not about the money, it’s about having the time. That said, the entire app is open source, so if you or anyone else would like to contribute to development, you're more than welcome!

Thanks again for your support and understanding.

Best regards,
Heszi

closed 2025-05-29 by heszegi

heszegi, 2025-06-02:
Try to create something like this:

[image]

[image]

heszegi, 2025-06-02:
The greater than / less than cards only trigger once - when the specified value is first reached, whether from a positive or negative direction. They won’t trigger again until the value crosses the threshold in the opposite direction and then returns.

heszegi, 2025-06-02:
On a typical sunny day, your Deye inverter should look something like this in Homey.

[image]

Iupiter1979, 2025-06-05:
Sorry, I’m not exactly sure what you mean by “data source” in this case. As far as I know, the homey app takes the data directly from Deye Cloud by linking the account, and there, all values (including solar production) appear correctly.

Is there anything else I should check?