heszegi/com.heszi.deye #22: AC loading management

kiteval11, 2025-09-29:
Do you have already been able to manage Batterie charging during night time with the grid ? 

In fact, just activating the card "set battery grid charges to On" is not enought, we need to change different parameters in "system work mode 2" :

we have to activate time of use

and modify the grid charge and the batt % 

Also on My deye station, I can't see the power from the solar cells. Is it normal 
My deye is a SUN-8K-SG05LP1-EU.

heszegi, 2025-09-30:
Hi kiteval11,

Please install and test the latest test version of the app and let me know how it works for you:
https://homey.app/en-hu/app/com.heszi.deye/Deye-Solar-Systems/test/

This version adds 2 new Flow Action cards: one to turn on/off TOU and another to set timeslot attributes.
Please note that the timeslot intervals can still only be managed through the official Deye app, as Homey flow cards don’t provide a suitable interface for that.

Best,
Heszi

kiteval11, 2025-10-01:
Hi Heszegi, 

Thanks a lot for this new feature, I'll test it right now. 

It's exactly what I needed that very nice.

For ce power cells I don't know here to check in the device settings ? also I have power from MPPT and also from micro-inverters via gen input.

kiteval11, 2025-10-01:
So I did some test, 

It's working but only if I set one timeslot at a time. If I want to modify 2 for exemple in the same flow, Only one of then take the modification, I have to send it a second time to have the second one wich take the modification.

Also if I modify another parameter (in the same flow) only one parameter it modified maybe it's a problem with deye which is not able to set parameters at the same time. I also tried with a time delay between the 2 timeslots (with advanced flows) 
- 30s delay between the Two cards activate only the second one same problem with 2min delay.

heszegi, 2025-10-02:
Hi, try the latest version 1.3.1: https://homey.app/en-hu/app/com.heszi.deye/Deye-Solar-Systems/test/

- timeslot input now supports multiselect, so you can set multiple slots at the same time (though likely only with the same value)
- added a 5s timeout after the API call to give the system a pause before the next flow card is triggered, this might help when chaining multiple cards

MPPTs and micro-inverters aren’t supported in the app, and since I don’t have any of these devices lying around, I’m not planning to add support myself, but the project’s open source, so if you’ve got the gear then feel free to jump in and contribute   😉 🚀⚡️

closed 2025-10-07 by heszegi